stooged/ESP32-Server-900u #105: Help with creating bin file

IRONB0SS, 2024-01-29:
Hey Sir, How can I compile/decompile esp32-s2 bin file?

Also please mention that Aurdino doesn't support customs tools since update 2.x in installation description

neo305, 2024-02-04:
it is impossible. I got lucky once and now after fresh install of Arduino I can't make bin files anymore.

stooged, 2024-02-04:
dont bother with 2.x use the legacy to compile

https://downloads.arduino.cc/arduino-1.8.19-windows.exe

[image]

closed 2024-02-04 by stooged

neo305, 2024-02-04:
Dude thank you so much. Yeah I've always used 1.8.19 and I never knew that was an option. I've always been told to use a bat file to build bin file. I'll try that option now thanks again for your time. I really appreciate it.

neo305, 2024-02-04:
am I doing something wrong? I u[pload and then sketch data upload like normal. then select what you said and get a 881kb bin file. should be 4mb
[image]

neo305, 2024-02-04:
[laz305 sketch upload.txt](https://github.com/stooged/ESP32-Server-900u/files/14155570/laz305.sketch.upload.txt)
here is my Arduino compiled sketch info

stooged, 2024-02-04:
the exported bin is not the same size as the full flash memory of the board, it will only contain the sketch/program code.

from laz305.sketch.upload.txt
`Sketch uses 902606 bytes (86%) of program storage space. Maximum is 1048576 bytes.`

your sketch uses 902606 bytes which will give you a file that shows about 881kb in windows.

neo305, 2024-02-04:
So there is no way to include the full flash memory of the board?

stooged, 2024-02-04:
not sure with the full flash as i dont export bin files, i just flash the board from the ide.