stooged/ESP32-Server-900u #111: Reading UART pins

Smig0l, 2024-02-11:
Can these ESP32 boards be used to [read](https://www.psxhax.com/threads/playstation-4-sac-001-ps4-uart-for-dummies-tutorial-e4-flasher-development.3623/) the UART pins on the motherboard?
Obviously at the same time they need to simulate a mass storage device and host a webserver and Access Point.

stooged, 2024-02-11:
yes you can read the uart pins by using [HardwareSerial](https://github.com/espressif/arduino-esp32/blob/master/cores/esp32/HardwareSerial.h) which is a part of the esp32 library.

you just need to setup the pins for rx / tx then set the baud rate on connect to 115200 and you will start to receive data.

you then need to send that data to something that can visualize it whether it is to the serial over usb or a wifi connection.

one of my early [simple esp servers](https://github.com/stooged/ESP32-Server/tree/master) had this functionality which would send the data over a wifi connection to port 24.

closed 2024-02-11 by stooged

Smig0l, 2024-02-12:
Thanks for the clarifications.
What board should I choose? There are thousands of them:
S2, S3, mini? Devkit C1? 8MB or 4MB

It seems they have the same capabilities

stooged, 2024-02-12:
they are all pretty much the same for this type of project.
more flash memory is good if you want to store more data on the board like payloads, html etc and have larger app partitions.
psram is another handy feature but you need to write the code to actually use it or it really does nothing.
dual core is a bit like psram where if you are not coding to actually use the second core it is really not doing much.

anything in the s2 or s3 versions is good.
i find the [feathers2](https://feathers2.io/) pretty solid for most projects it has 16mb flash, single core and has 8mb of psram.
they do a [feathers3](https://esp32s3.com/feathers3.html) now which is much the same but with the dualcore but i have not tried one yet.

the s3 board i test with is the [ESP32-S3-DevKitC-1-N8R8](https://docs.espressif.com/projects/esp-idf/en/stable/esp32s3/hw-reference/esp32s3/user-guide-devkitc-1.html#description-of-components).
this one is handy for the ps4 exploit because it has 2 usb ports so it does allow you to plug one port into the ps4 for exfathax and the other port into a pc for serial debugging at the same time.

so if you were looking to read uart data from the ps4 you could just [pass the data to the serial](https://github.com/stooged/ESP32-Server/blob/10a20d227b06121480edf96bdc3426ffc889afba/ESP32_Server/ESP32_Server.ino#L171) and read it via usb.