stooged/ESP32-Server-900u #112: Exception using psfree

jacoghi, 2024-02-28:
Default settings, with the following changes:
https://github.com/jacoghi/ESP32-Server-900u/commit/6854ecd9895c4ec7bc52d126d7ec88b8e4f0e4ee

Tried some 10 times, rebooted and tried again some 10 times, keep getting this, all the time. Known issue with psfree? Iḿ unsure of it cause as it can be seen from the background, the server actually reads: "Loading USB ExFatHax", so I was assuming the psfree part of the hack had already succeeded.
[image]
Thanks!

jacoghi, 2024-02-28:
Just tried disabling psfree. It now doesn´t give me an exception messages but the jailbreak never succeeds either, keep getting the failed to trigger exploit message. Will try reverting completely to what we had before this year updates

stooged, 2024-02-29:
[image]

make sure all of those are disabled in the board settings as on some boards they mess with the exfathax process.

jacoghi, 2024-02-29:
Reset hard till 2022, changed this USB CDC ON boot which was enabled to disabled, reverted libraries to their git master versions instead of the updated versions that arduino ide installs, now it's working as before. Will roll back to 2024 with psfree and try again to narrow it down.
Thanks man

jacoghi, 2024-02-29:
Reverted everything to thje most updated versions, culprit was this USB CDC On Boot which was enabled by Default on Arduino IDE 2.3.2. Lolin S2 Mini board here

closed 2024-02-29 by jacoghi

neo305, 2024-02-29:
stooged can you tell me please how do I update firmware to 1.05?[image]

stooged, 2024-02-29:
> stooged can you tell me please how do I update firmware to [...]

Thats a modification to my original code so i dont know what they have done.
It would be best to contact the person releasing that and ask them.