stooged/ESP32-Server-900u #114: I bought the recommended dongle from Aliexpress - not recognized by Windows 10

Zoukambada, 2024-04-16:
I bought [my dongle](https://aliexpress.com/item/1005005387342730.html) that was pre-installed with [Kameleon](https://github.com/kmeps4/Kme900_Auto_ESP32_S2/) and I entered the Admin and these are the info that I got:

`###### Software ######

Firmware version 1.00
SDK version: v4.4.3
###### Board ######

MCU: ESP32S2
Board: QT Py ESP32-S2
Chip Id: ESP32-S2
CPU frequency: 240MHz
Cores: 1
###### Flash chip information ######

Flash chip Id: 0
Estimated Flash size: 4.00 MB
Flash frequency: 80.00 MHz
Flash write mode: QIO
###### Storage information ######

Filesystem: FatFs
Total Size: 2.85 MB
Used Space: 1.36 MB
Free Space: 1.49 MB
###### Ram information ######

Ram size: 187.32 KB
Free ram: 43.58 KB
Max alloc ram: 21.49 KB
###### Sketch information ######

Sketch hash: 7eda197353d7f6072a08859944ce617e
Sketch size: 872.78 KB
Free space available: 151.22 KB`

Then I discovered that it might be the Adafruit QT Py ESP32-S2.

In my Windows 10 I went to Device Manager, something was already in COM1 port. Then I inserted the dongle and nothing new appeared.
Then I follow these steps:
- I installed the [USB to UART Bridge](https://www.silabs.com/developers/usb-to-uart-bridge-vcp-drivers) - I tried CP210x_Universal_Windows_Driver.zip and CP210x_VCP_Windows.zip. Both times I rebooted my PC and it still wasn't recognized in Device Manager.
- I installed Ardruino IDE, installed the Library Adafruit QT Py ESP32-S2 following this [guide here](https://learn.adafruit.com/adafruit-qt-py-esp32-s2/arduino-ide-setup).
- I installed your required libraries for Ardruino IDE: [ESPAsyncWebServer](https://github.com/me-no-dev/ESPAsyncWebServer) and [AsyncTCP](https://github.com/me-no-dev/AsyncTCP)
- Ardruino IDE still says that the dongle is not connected (in the status) 
[image].
- I also tried NodeMCU-PyFlasher.exe with no success (not recognized).

I watched Modded Warfare youtube [ESP32-S2 guide](https://www.youtube.com/watch?v=Cllj6yDha6o) and also the guide from [Michael Crump](https://www.youtube.com/watch?v=FteGc33nmAg) and I realized that their dongles have 2 buttons and they say it needs to be pressed. Te thing is, mine is the same as the recommended one here in this Github so I really don't know what to do. I pressed it once and hold it and nothing new happened.
 
[image]

Then I tought that maybe one of the soldering points needs to be redone. I dunno if it could be the problem. 
[image]

Any toughts?

neo305, 2024-04-16:
I got the same one. I still haven't figured it out. Try holding the button down while plugging it in, which is impossible I know.

stooged, 2024-04-16:
the cdc mode of that dongle is disabled so you need to hold the little button down with a pin while you plug it in and release it after it is plugged in.
it will then show up as a numbered com port to the system so you can flash it.

closed 2024-04-16 by stooged

Zoukambada, 2024-04-16:
@neo305 I did what @stooged told and it worked! Now it's appearing in Device Manager as seen below:
[image]

Thanks!