stooged/ESP32-Server-900u #115: PPPoE over wifi AP?

Smig0l, 2024-05-07:
I know is not relevant for the exploit in this repo.
But can the esp32 (without a Ethernet module) works as a PPPoE server using a wifi access point?
Searching for it leads only to PPPoS (is not the same right?)

stooged, 2024-05-07:
the ps4 has no option for pppoe over wifi and the pppwn exploit is rather tricky to execute so it is not really going to be something for esp boards.

it will something more for a raspberry pi which you can pick up very cheap these days.

i am only focusing on a pi solution for pppwn.

closed 2024-05-07 by stooged

Smig0l, 2024-05-08:
yes as I imagined. I also have a couple of Raspberry lying around. I await another awesome project from you.