stooged/ESP32-Server-900u #116: Feature Request: Switching off the ESP32-S2/S3 module immediately after loading the GoldHEN

muxi1, 2024-06-03:
It would be good to include a way of shutting down the ESP32-S2/S3 immediately after loading the GoldHEN so that it does not get too hot due to prolonged operation.

stooged, 2024-06-03:
You can set the time to sleep value so the esp shuts down after a certain amount of time to prevent prolonged operation.

https://github.com/stooged/ESP32-Server-900u/blob/main/ESP32_Server_900u/ESP32_Server_900u.ino#L87

closed 2024-06-03 by stooged

neo305, 2024-06-04:
Yes but he is talking about a esp32-s3 with a lcd screen.

stooged, 2024-06-04:
> Yes but he is talking about a esp32-s3 with a lcd screen.

There is no support for a lcd screen in this project so it would remain unused.
the best way to tackle that is to workout which gpio controls the backlight and switch it off in the setup function of the code so it is always off.

muxi1, 2024-06-07:
> Yes but he is talking about a esp32-s3 with a lcd screen.

That's exactly what I was getting at!

stooged, 2024-06-07:
> > Yes but he is talking about a esp32-s3 with a lcd screen.
> 
> That's exactly what I was getting at!

Just workout which gpio controls the backlight and set it to be off when the esp boots in [setup()](https://github.com/stooged/ESP32-Server-900u/blob/main/ESP32_Server_900u/ESP32_Server_900u.ino#L621) 
You are not using the lcd for anything so there is no point having it on at all.

muxi1, 2024-06-09:
> Just workout which gpio controls the backlight and set it to be off when the esp boots in [...]

I don't know how to do this, as I have hardly any programming knowledge. I think I will buy a LilyGo T-Dongle without LCD display. Thanks for your answers anyway.

stooged, 2024-06-09:
> I don't know how to do this, as I have hardly any programming knowledge. I think I will buy a LilyGo T-Dongle without LCD display. Thanks for your answers anyway.

now that you have mentioned a specific board i will assume you have the t-dongle with the lcd and the gpio to disable the backlight is 38 high.
i have updated the repo to pull that gpio high on boot and keep it that way during the sleep.

you can also use this repo for the t-dongle, it is setup specifically for that board and the use of a sd card to store the payloads.
https://github.com/stooged/PS4-Dongle

muxi1, 2024-06-10:
> now that you have mentioned a specific board i will assume you have the t-dongle with the lcd and the gpio to disable the backlight is 38 high. i have updated the repo to pull that gpio high on boot and keep it that way during the sleep.
> 
Thank you very much for that update.