stooged/ESP32-Server-900u #118: unable to compile sketch

iguanatamer, 2024-06-14:
Hi,

I get an error compiling the sketch.

C:\Users\admin\Documents\ESP32-Server-900u-main\ESP32_Server_900u\ESP32_Server_900u.ino: In function 'void setup()':
C:\Users\admin\Documents\ESP32-Server-900u-main\ESP32_Server_900u\ESP32_Server_900u.ino:761:10: error: 'class WiFiClass' has no member named 'setAutoConnect'; did you mean 'setAutoReconnect'?
  761 |     WiFi.setAutoConnect(true);
      |          ^~~~~~~~~~~~~~
      |          setAutoReconnect
C:\Users\admin\Documents\ESP32-Server-900u-main\ESP32_Server_900u\ESP32_Server_900u.ino: In function 'void loop()':
C:\Users\admin\Documents\ESP32-Server-900u-main\ESP32_Server_900u\ESP32_Server_900u.ino:1045:7: error: 'gpio_hold_en' was not declared in this scope; did you mean 'gpio_mode_t'?
 1045 |       gpio_hold_en((gpio_num_t)38);
      |       ^~~~~~~~~~~~
      |       gpio_mode_t
C:\Users\admin\Documents\ESP32-Server-900u-main\ESP32_Server_900u\ESP32_Server_900u.ino:1046:7: error: 'gpio_deep_sleep_hold_en' was not declared in this scope
 1046 |       gpio_deep_sleep_hold_en();
      |       ^~~~~~~~~~~~~~~~~~~~~~~
Multiple libraries were found for "WiFi.h"
  Used: C:\Users\admin\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.1\libraries\WiFi
  Not used: C:\Users\admin\Documents\Arduino\libraries\WiFi
exit status 1

Compilation error: 'class WiFiClass' has no member named 'setAutoConnect'; did you mean 'setAutoReconnect'?

eemcm2, 2024-06-14:
Same problem here

eemcm2, 2024-06-14:
> You need to use Espressif core V2 (2.0.17) to build this sketch. Recent V3 update has brought with it a lot of major API changes that break the build.

Thanks! Problem solved

stooged, 2024-06-14:
install the esp32 [core 2.0.14](https://github.com/espressif/arduino-esp32/releases/tag/2.0.14) as it seems to be the last one that does not break too much.
esp-idf 5+ has broken many libraries.

closed 2024-06-14 by stooged