stooged/ESP32-Server-900u #122: Compile error just started to happen

Tribb, 2024-07-16:
not sure why this started to happen but Im getting this error  
Arduino: 1.8.19 (Windows 10), Board: "LOLIN S2 Mini, Disabled, Disabled, Disabled, No OTA (1MB APP/3MB FATFS), None, Enabled"
C:\Users\tribb\Desktop\ESP32-Server-900u-main\ESP32_Server_900u\ESP32_Server_900u.ino: In function 'void setup()':
ESP32_Server_900u:761:10: error: 'class WiFiClass' has no member named 'setAutoConnect'; did you mean 'setAutoReconnect'?
  761 |     WiFi.setAutoConnect(true);
      |          ^~~~~~~~~~~~~~
      |          setAutoReconnect
C:\Users\tribb\Desktop\ESP32-Server-900u-main\ESP32_Server_900u\ESP32_Server_900u.ino: In function 'void loop()':
ESP32_Server_900u:1045:7: error: 'gpio_hold_en' was not declared in this scope; did you mean 'gpio_mode_t'?
 1045 |       gpio_hold_en((gpio_num_t)38);
      |       ^~~~~~~~~~~~
      |       gpio_mode_t
ESP32_Server_900u:1046:7: error: 'gpio_deep_sleep_hold_en' was not declared in this scope
 1046 |       gpio_deep_sleep_hold_en();
      |       ^~~~~~~~~~~~~~~~~~~~~~~
Multiple libraries were found for "WiFi.h"

 Used: C:\Users\tribb\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.2\libraries\WiFi
 Not used: D:\Arduino\libraries\WiFi
Multiple libraries were found for "AsyncTCP.h"
 Used: C:\Users\tribb\OneDrive\Documents\Arduino\libraries\AsyncTCP
 Not used: C:\Users\tribb\OneDrive\Documents\Arduino\libraries\AsyncTCP-master
exit status 1
'class WiFiClass' has no member named 'setAutoConnect'; did you mean 'setAutoReconnect'?

stooged, 2024-07-16:
install the esp32 [core 2.0.14](https://github.com/espressif/arduino-esp32/releases/tag/2.0.14) as it seems to be the last one that does not break too much.
esp-idf 5+ has broken many libraries.

closed 2024-07-16 by stooged

Tribb, 2024-07-17:
Thankyou for a quick fix and thanks for all you have done inc app2usb which i still use