stooged/ESP32-Server-900u #124: ESP32_Server_900u.ino:761:10: error: 'class WiFiClass' has no member

Encapsulate, 2024-09-09:
Arduino IDE 2.3.2, Libraries: AsyncTCP (by dvarrel) v1.1.1, ESP Async Webserver (by Me-No-Dev) v1.2.4, ESP32S2 in FDU mode, Additional boards manager URL: https://raw.githubusercontent.com/espressif/arduino-esp32/gh-pages/package_esp32_index.json,
board: ESP32S2 Dev Module, COM7, Internal USB, PSRAM Enabled, 

Results of Uploading/Compiling sketch...
Line 761 of ESP32_Server_900u.ino <ERROR>

`
C:\Users\x\Desktop\PS4 DEV\ESP32-S2 Jailbreak\ESP32-Server-900u-main\ESP32-Server-900u-main\ESP32_Server_900u\ESP32_Server_900u.ino: In function 'void setup()':
C:\Users\x\Desktop\PS4 DEV\ESP32-S2 Jailbreak\ESP32-Server-900u-main\ESP32-Server-900u-main\ESP32_Server_900u\ESP32_Server_900u.ino:761:10: error: 'class WiFiClass' has no member named 'setAutoConnect'; did you mean 'setAutoReconnect'?
  761 |     WiFi.setAutoConnect(true);
      |          ^~~~~~~~~~~~~~
      |          setAutoReconnect

exit status 1

Compilation error: 'class WiFiClass' has no member named 'setAutoConnect'; did you mean 'setAutoReconnect'?
`
To fix, I had to remove line 761.

vasmani, 2024-10-12:
@Encapsulate 
[image]

I installed ESP32 version 2.0.14 and it worked!

Leerz, 2024-12-20:
> @Encapsulate <img alt="Screenshot 2024-10-12 at 5 16 10 PM" width="305" src="https://private-user-images.githubusercontent.com/1195146/375939943-ff919371-c104-4df6-9df8-5c526e51cd3c.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzQ2ODc0NzUsIm5iZiI6MTczNDY4NzE3NSwicGF0aCI6Ii8xMTk1MTQ2LzM3NTkzOTk0My1mZjkxOTM3MS1jMTA0LTRkZjYtOWRmOC01YzUyNmU1MWNkM2MucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MTIyMCUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDEyMjBUMDkzMjU1WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9ZWVjZTE0YTA2ZmVkZGYwMGZhYmNmMjA2NWQwNGI1NWQ1YTkyMjEzZmIwODExNDNmYmRiOTU0ZTYxZTliNjU5YSZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QifQ.uKOFPsc9wFFMFwTl7GYc7cf2MN12lEsSfY1dhox6y6g">
> 
> I installed ESP32 version 2.0.14 and it worked!

20241220 - works