stooged/ESP32-Server-900u #125: Error

Hukishi, 2025-01-12:
"Sketch uses 1348175 bytes (102%) of program storage space. Maximum is 1310720 bytes.
Global variables use 63452 bytes (19%) of dynamic memory, leaving 264228 bytes for local variables. Maximum is 327680 bytes.
Sketch too big; see https://support.arduino.cc/hc/en-us/articles/360013825179 for tips on reducing it.
text section exceeds available space in board

Compilation error: text section exceeds available space in board"

I have this error and I don't know how to solve it

Hukishi, 2025-01-12:
I had version 3.1.1 installed in the board manager and when I downgraded it to 2.0.2 I managed to compile

dexter85, 2025-02-05:
I have the same board, and it works with ESP32 2.0.17. With ESP32 3.1.1, I have the same issues.